cmm-21/a2 #1: Why kinematic walking controller fails in real world?

eastskykang, 2021-03-18:
Please watch this video: https://youtu.be/PfeJYcg-yk0

and feel free to discuss with your colleagues!

FenglongSong, 2021-03-18:
I think the reason why the robot jumps in the video is: 
you can see at the beginning of the video, the robot rotates around the diagonal line of two diagonal stancing feet due to gravity. Then one of the swing feet gets closer to the ground. When touching the ground, this foot hasn't reached the lowest point in pre-planned trajectory yet. Since only kinematics are considered, the motion of the whole leg is very stiff (no compliance). This foot keeps exerting force onto the ground, which causes the robot to jump.

The kinematic walking controller is totally open-loop. I guess it will work only under very ideal circumstances, e.g. perfect initial conditions, totally flat ground and no disturbance.

eastskykang, 2021-03-19:
@FenglongSong This is indeed a brilliant observation! BTW, can you elaborate more on "The kinematic walking controller is totally open-loop" ?

schmirob, 2021-03-19:
This is interesting, my thoughts are if the robot is just controlled kinematically the causes of these motions are not considered, i.e. the contact forces. Since in kinematic control it is not measuring its joint torques or a similar measurement with which the contact forces can be estimated there is no feedback on what the current contact forces are so it is a sense working blindly / open loop. Further the current gait pattern, where only 2 feet are on the ground a the time, is statically unstable. Performing this gait pattern needs to constantly stabilize the system so feedback is required.

FenglongSong, 2021-03-19:
@eastskykang  Thank you. As Robin said @schmirob , the main reason that the kinematic controller doesn't work for trotting gait is the robot is statically stable. I believe a kinematic controller may have slightly better performance for a crawling gait, since having 3 stancing feet can bring statically stability. But I think if the kinematic controller is deployed on a real robot, it will somehow perform better than in the simulation, because the joint of a real robot always have some compliance. It is not possible for a real robot leg to produce a very huge support force at footip as in the simulation. So the possibility for jumping and then falling down will be reduced to some extent.

A less rigorous opinion: the control strategies of quadruped and quadrotor are actually similar. Four feet are just like four rotors, which can exert force and torque to control roll, pitch and yaw of the floating base. (The difference is, for quadrotors, the directions of propelling forces are fixed and the magnitude changed continuously. But for quadruped, the directions can change and magnitude is discrete since legs switch between stance and swing). Both of them are statically unstable. Without feedback control, the force generated by propellers or feet cannot stabilize the system.

eastskykang, 2021-03-20:
@schmirob @FenglongSong brilliant! 

Yes, both quadrupedal robot and quadrotor are floating base. The main difference is (in general) quadrotor is always under-actuated while quadrupedal robot can be in under-actuated configuration, over-constrained configuration depending on its configuration. Also quadrupedal control problem is "hybrid" control problem since control scheme for contact phase and swing phase should be different. Finally quadrotor model is very simple. You can easily model quadrotor's dynamics as a linear system while quadrupedal, it's not the case. 

By the way, it's open-loop in the sense, there's no way to control base pose unless you take into account the ground reaction force in your controller. But in joint level, (if we use PD control for each joints as in this joint) I'd say it's a closed-loop controller in joint-level.

FenglongSong, 2021-03-20:
Cool, that's much more precise!

eastskykang, 2021-03-20:
@FenglongSong 

> "it will somehow perform better than in the simulation, because the joint of a real robot always have some compliance. It is not possible for a real robot leg to produce a very huge support force at [...]

I agree that a bit of compliance in the joints would be helpful. The thing is, we don't really know how "compliant" it should be. This is indeed the caveat of kinematic level control.

PanagiotisAnagnostaras, 2021-03-20:
In real world I believe that a kinematic controller cannot exist since our controllable variables are the torques of the motors, or, if we do a step back, their voltage. However if we know the desired end effector trajectories and velocities we can calculate the desired joint position and velocity with inverse kinematics and calculate the torque at each joint with a PD controller in relation with the errors in q and q_dot (which as far as I understand is implemented at the video). 

If this is the case, then we need feedforward terms in order to compensate for the nominal/ expected/ modelled dynamics and let feedback to take care of the unmodelled dynamics and disturbances. Moreover the reason why the robot jumps I think that it may be explained from the way that the contacts has been simulated. For example if we consider soft  contacts as a spring-damper system, then we could change the parameters in order to have a more-realistic behaviour but then for my perspective the question which arises is whether we use simulations and computer aided engineering in order to "predict" a real world scenario or we "tune" the simulation in order to have a realistic outcome.

eastskykang, 2021-03-20:
@PanagiotisAnagnostaras You guys do have very good understanding of legged robotics! Yes, indeed, here the key is the ground reaction forces from contacts. In our kinematic controller (and PD controller), we don't address them. That's why it's very easy to fail. As soon as there's an interaction between robot and environment, kinematic level control is not enough. 

let me leave some comments here:

```
since our controllable variables are the torques of the motors, or, if we do a step back, their voltage
```

in fact, this is rather motor *current* targets because motor torque output is related to motor current. There're some other factors such as gear ration etc. should be considered to.  

```
we need feedforward terms in order to compensate for the nominal/ expected/ modelled dynamics
```

Yes. In fact, this is the standard way of legged locomotion control, which is called "model-based control" approach. Instead of inverse kinematics, we solve *inverse dynamics* problem and do *force control* (we compute desired-joint torque as well as corresponding ground reaction force that allows a robot to follow the targets)

```
and let feedback to take care of the unmodelled dynamics and disturbances.
```

This scheme is called *impedance control*. We give a target feedforward torque as well as feedback targets (joint angle + joint speed). This allows more safe interaction with the "environment". 

```
For example if we consider soft contacts as a spring-damper system, then we could change the parameters in order to have a more-realistic behaviour but then for my perspective the question which arises is whether we use simulations and computer aided engineering in order to "predict" a real world scenario or we "tune" the simulation in order to have a realistic outcome.
```

Definitely. I'd say "predict a real world scenario" is more close to what model-based control does, and commonly used for legged locomotion control. Meanwhile "we "tune" the simulation" is related to research keywords "real-to-sim" or "domain adaptation".

eastskykang, 2021-03-20:
@FenglongSong 

BTW, I found this is also pretty interesting 

>  the main reason that the kinematic controller doesn't work for trotting gait is the robot is statically stable. I believe a kinematic controller may have slightly better performance for a crawling gait, since having 3 stancing feet can bring statically stability.

Precisely. But if you watch the YouTube videos of legged robot demo, they rather show "trotting" gait than "crawling" gait. Also in it's standard walking mode, most commercial legged robots are doing "trotting". Can you make a guess why?

FenglongSong, 2021-03-21:
I guess it's because trotting gait gives both suitable speed and stability for commercial legged robots. Crawling is usually slow but it produces better stability than trotting. I think commercial quadrupedal robots sometimes use crawling gait to pass through some very uneven terrains, because these tasks have a higher demand on stability. As for other high-speed gaits like gallop, they're usually not used since we seldom need legged robots to run that fast in current commercial use. Also, these gaits might have higher requirements on hardware, which increases the cost.

Hustwireless, 2021-03-25:
> @FenglongSong
> 
> > "it will somehow perform better than in the simulation, because the joint of a real robot always have some compliance. It is not possible for a real robot leg to produce a very huge support force at footip as in the simulation. So the possibility for jumping and then falling down will be reduced to some extent."
> 
> I agree that a bit of compliance in the joints would be helpful. The thing is, we don't really know how "compliant" it should be. For instance, if the PD gain is too low, this happens.
> 
> https://www.youtube.com/watch?v=pAdHV_vNN4g
> 
> This is indeed the caveat of kinematic level control.

Does a low PD gain inplies too much compliance?

eastskykang, 2021-03-25:
@Hustwireless Exactly. If control gain (specifically P gain) is too low, trajectory tracking performance would be very poor and this can lead the robot to fall down. 

So you need to find a *sweet spot* of control gain. There are some guidelines for tuning these value but rather we use another control strategy for more robust locomotion control. 

Hope this helps!

eastskykang, 2021-03-30:
https://youtu.be/8A5-u_9enJ0

Please find this video if you are interested in relevant topics.