cmm-21/a2 #4: locomotion.exe closed as long as I hit the play button

breadli428, 2021-03-18:
Hi,

I am trying to verify my code for task 2.2, however the locomotion.exe will be immediately closed as long as I hit the play button. It seems to be a sign that my coding is somehow wrong, but I am not sure the reason behind.

Plus, the ground and the feet of the robot do not align, and this it not an individual case.

Thanks!

eastskykang, 2021-03-19:
@breadli428 

please see #6 issue for feet-under-the-ground. 

If the app crashes, please change your cmake mode into Debug (you can do this as follows)

[image]

and click the "launch debugger" (the button with "a bug" shaped icon). This will give you more rich information. (it will stop at the point where the code crashes)

breadli428, 2021-03-19:
Hi Dongho,

Thanks I guess I found the problem. But I have a follow up question.

In function solve in IK_Solver.h, how is nSteps = 10 defined? Where is the time information in endEffectorTargets stored?

Thanks!

closed 2021-03-19 by breadli428