cmm-21/a2 #5: Failed to load material file

Catially, 2021-03-19:
Dear TAs, 

I tried to run locomotion app, but it shows this:

Material file [ /home/jin/git/a2-Catially/data/meshes/grid1.mtl ] not found.
Failed to load material file(s). Use default material.

Material file [ /home/jin/git/a2-Catially/data/meshes/grid2.mtl ] not found.
Failed to load material file(s). Use default material.

Both `d` and `Tr` parameters defined for "Default". Use the value of `d` for dissolve (line 8 in .mtl.)

no diffuse texture found for shape `Default`
Segmentation fault (core dumped)

I am not sure about how should I solve this. Can you help me?

eastskykang, 2021-03-19:
Can you play the app? If that's the case, it's good to ignore the message.

Catially, 2021-03-19:
Sorry, no. The app crashed right after all the error messages.

Catially, 2021-03-19:
I guess it is not the material file load error because I tried to reconstruct everything but had the same problem. I will check my code again.

closed 2021-03-19 by Catially