cmm-21/a2 #7: Release mode perfect but debug mode failed

breadli428, 2021-03-19:
Hi,

I finished ex 2.2 and it worked perfectly under release mode, however the green ball indicators as well as the feet do not move (trot) when I run it under the debug mode.

Is this normal?

Thanks!

eastskykang, 2021-03-19:
@breadli428 Hi,

Does the app freeze (and don't respond at all)?

breadli428, 2021-03-19:
Not freeze. I can click the play button and the yellow dots depicting the feet trajectories moves as normal. But the green dots and the feet won’t move at all.

eastskykang, 2021-03-20:
I will test this on my Windows machine and come back as soon as possible.

eastskykang, 2021-03-20:
Unfortunately, I couldn't reproduce this problem on my Windows machine. Can we discuss this problem in the next tutorial session (on 25th of March)?

breadli428, 2021-03-20:
Sure! Thanks Dongho!

eastskykang, 2021-03-29:
#33 @breadli428 can you quickly try this and let me know if this issue is still there or not?

eastskykang, 2021-03-31:
@breadli428 awesome! It's super normal to be slow in debug mode :)

I will close this issue now.

closed 2021-03-31 by eastskykang