cmm-21/a2 #8: Foot trajectory planning unstable

Catially, 2021-03-19:
Dear TA,

I managed to finish exercise 3, but when I tried to increase turning speed, the planned foot trajectory is very unstable, which results in the unstable motion of the feet. 

When I only changed the turning speed to 0.1, the robot would start to "dancing". In the picture below, we can clearly see that there are some trajectories in the circle (stable trajectory) or even out of the circle. And this also results in the instability of IK.

Is this supposed to be solved by us, or we can just leave it?

[image]

breadli428, 2021-03-19:
Same problem here, not sure it is a technical problem or out of instability.

eastskykang, 2021-03-20:
This looks a bit suspicious, it shouldn't be the case. As you can see from the example video, trajectory planner should be stable enough to make turning motion: https://youtu.be/NwVNkggKhC0

breadli428, 2021-03-20:
I used Newton-Gauss and did not yet implement line search and regularization. Does this result in the instability?

breadli428, 2021-03-20:
Ah, now I don't think it's because of the estimation error of the Jacobian, since I implemented the analytical one and it's still "dancing".

eastskykang, 2021-03-20:
From the screenshot you posted here, it could be the case that your forward kinematics implementation does not update position of ee well if the base orientation is not 0, 0, 0. 

The base orientation (qIndex 3, 4, 5) components of generalized coordinate are euler angles. One thing you should keep in mind is that in our case, we use Z(roll) X(pitch) Y(yaw) euler angles, since forward direction of the robot is z axis, and sideways is x axis and world-up is y axis.

Catially, 2021-03-21:
Problem identified.

Both my FK and Locomotion plan have some mistakes. It is not a technical issue. Thanks, Dongho.

closed 2021-03-21 by Catially