cmm-21/a2 #9: where to find terrain file

breadli428, 2021-03-20:
Hi,

I was trying to do Ex 5, however I had problem locating the terrain file. I tried to search but a lot of similar result came out and I did not manage to find the code where it loads the terrain.

Could you please give more information on this?

Thanks!

eastskykang, 2021-03-20:
You can create your own terrain by adding some large spheres in the scene as I mentioned earlier. 

If you want to use heightmap, try one in the data folder. But, loading terrain.obj file in the scene would require some extra work.

breadli428, 2021-03-20:
-"You can create your own terrain by adding some large spheres in the scene as I mentioned earlier."

Yes, but where, actually? In terrain.obj?

eastskykang, 2021-03-22:
@breadli428 I'm closing this issue. Please reopen once you encounter another problem.

closed 2021-03-22 by eastskykang