cmm-21/a2 #10: Play unstable

Glanfaloth, 2021-03-21:
For ex.2, the trotting sometimes works fine. But the app window sometimes closes after a few seconds, sometimes closes immediately after clicking play button. Any idea why? Thank you! (I am using Gauss Newton)

Glanfaloth, 2021-03-21:
The jacobian matrix turns nan nan nan... at some point, what could be the reason

eastskykang, 2021-03-22:
Could be two reasons. 
1. wrong Jacobian matrix. 
2. numerical unstability. 

Gauss-Newton update includes (**J**<sup>T</sup>**J**)<sup>-1</sup>**J**<sup>T</sup> term. Here, computing inverse matrix by ```.inverse()``` is never a good idea. 

Rather use linear solver as I suggested in the hint:

```cpp
// - when you compute inverse of the matrix, use ldlt().solve() instead of inverse() function. this is numerically more stable.
//   see https://eigen.tuxfamily.org/dox-devel/group__LeastSquares.html
```

Glanfaloth, 2021-03-22:
Thank you Dongho, I did use ldlt().solve() for inversion. Could the Jacobian matrix be wrong even when the doggy trotts perfectly before the app crashes?

eastskykang, 2021-03-22:
@Glanfaloth I think I need more information. Could we discuss about this on Thursday's tutorial session? Meanwhile, you may want to try this in debugging mode to identify what causes the problem. In debug mode, eigen library will give you more rich information that help you to find the problem. (if the app crashes, debugger will stop at the exact point causes the problem)

Glanfaloth, 2021-03-25:
Debugging tells me "Assertion failed: (IS_ZERO( (getRotationQuaternion(gamma, c) * getRotationQuaternion(beta, b) * getRotationQuaternion(alpha, a) * q.inverse()) .vec() .norm() / 10e5)), function computeEulerAnglesFromQuaternion, file ./src/libs/utils/include/utils/mathUtils.h, line 308."
I don't understand...

eastskykang, 2021-03-25:
This indicates the robot's state value were diverged. 

I think we should identify when the Jacobian exactly turns to nan. I'd like to suggest you to simulate the IK controller with the button here. It advances just one step of simulation. Print out Jacobian matrix, and see how the value changes as you do the simulation. 

[image]

Also please record your simulation here. That would help to discuss.

Glanfaloth, 2021-03-25:
I already tried printing out Jacobian matrix. The results differ from time to time. Sometimes it turns nan immediately and sometimes the it lasts for one minute before crashing. Averagely things go wrong after around 10 sec.
Here is a screenshot of the moment when the Jacobian changes to nan:
[image]

eastskykang, 2021-03-25:
@Glanfaloth I quickly reviewed your IK solver implementation. Please try not overwriting Jacobian matrix but rather declare another matrix variable and let me know if it works.

Glanfaloth, 2021-03-25:
Wow it works now!!! Thank you so much!!!!

closed 2021-03-25 by Glanfaloth