cmm-21/a2 #12: forward direction

leopoldsedev, 2021-03-21:


eastskykang, 2021-03-22:
@leopoldsedev hello, 

```robot->forward``` vector is a vector expressed in robot's bframe, not world frame. Thus, you have to take into account robot's orientation to correctly generate future trajectory. 

 ```cpp
Quaternion heading =
                getRotationQuaternion(headingAngle, V3D(0, 1, 0));
```

at the top of while loop block compute the orientation of the base. You can use this quaternion variable. 

Note. do not update ```robot->forward```. This is deliberately chosen to be expressed in the bframe.

iqiac, 2021-03-23:
Sorry, I'm unfamiliar with quaternions. How are we supposed to use this to change the moving direction  of the robot? Is any comments in the code or theory on the slides? My robot also moves only in one direction but it can "turn" resp. spin, although the legs start to get fancy if I spin it.

eastskykang, 2021-03-23:
@iqiac I cannot say too much details because it could be directly related to the task. Please take a look this diagram. Your task is incrementing position and heading angle accordingly.

[image]

Quaternion is nothing but rotation representation. Here ```Quaternion heading  = getRotationQuaternion(headingAngle, V3D(0, 1, 0));``` is the (future) orientation of the robot's base with respect to the world frame. More formally, this can be expressed as **R**<sub>WB</sub>

Thus, you can get forward direction vector of the robot expressed in the world frame **v**<sub>W</sub> by **v**<sub>W</sub> = **R**<sub>WB</sub> **v**<sub>B</sub>

Hope this helps.

eastskykang, 2021-03-23:
p.s. P3D is a position and V3D is a 3D vector: vector is difference of two positions i.e.

```
P3D start = ...;
V3D vectorFromStartToEnd = ...;
P3D end = start + vectorFromStartToEnd;
```

iqiac, 2021-03-24:
Ok thanks you, that helped for the turning.

closed 2021-03-24 by eastskykang