cmm-21/a2 #13: Analytic Jacobian: passed the test on my computer but failed it in GitHub

Catially, 2021-03-22:
Dear TA,

I encountered a strange issue. My implementation of analytic jacobian passed the test-a2 on my computer, while it failed the GitHub test. Besides that, I tried to use the analytic Jacobian in test 2.2, the robot will not moving as if I didn't write any code in the IK_solver.h. Also, I noticed that there are some Cmake errors in the GitHub auto-grading, do they cause these issues?

I am not sure whether there is a difference between tests on my computer and on GitHub. Maybe could you please help me with it?

[image]
[image]

eastskykang, 2021-03-22:
1. github action is currently failing. Please see #3. You can just run the test by yourself 
2. We need a more information. Obviously, if the test passes and your IK solver work properly with Jacobian matrix computed by FD, this should work without any problem.

Catially, 2021-03-22:
I tried to test it as follows:

This is a small part of IK_solver.h. I tried to calculate analytic jacobian first and overwrite it with estimate jacobian. And this code is working perfectly. 
```
gcrr.compute_dpdq(endEffectorTargets[j].p, endEffectorTargets[j].rb, dpdq); //analytic jacobian
std::cout<<"analytic done!"<<std::endl; 
gcrr.estimate_linear_jacobian(endEffectorTargets[j].p, endEffectorTargets[j].rb, dpdq); //estimate jacobian
std::cout<<"estimate done!"<<std::endl;
```

But if I changed the order, which is
```
gcrr.estimate_linear_jacobian(endEffectorTargets[j].p, endEffectorTargets[j].rb, dpdq); //estimate jacobian
std::cout<<"estimate done!"<<std::endl;
gcrr.compute_dpdq(endEffectorTargets[j].p, endEffectorTargets[j].rb, dpdq); //analytic jacobian
std::cout<<"analytic done!"<<std::endl; 
```
This will result in the same problem I had before. 
And it showed that the calculation is successfully carried out.
[image]

eastskykang, 2021-03-22:
can you use different matrix variable? 

what happens if 

```cpp
Matrix dpdq2;
gcrr.compute_dpdq(endEffectorTargets[j].p, endEffectorTargets[j].rb, dpdq2);

and use ```dpdq2``` for IK?

Catially, 2021-03-22:
Still have the same issue. 
I just wonder maybe because my analytical Jacobian calculation is too slow?

eastskykang, 2021-03-22:
If your analytical Jacobian implementation is correct, it should be much faster than numerical one. However, I don't believe this is relevant to the computation speed. 

Please try to print out the value of Jacobians and see if there's any suspicious value. Also, do a double check if your IK solver works properly with numerical Jacobian implementation.

Catially, 2021-03-22:
I tried another way of implementation, passed the test as well. And my dog finally started to move, but unfortunately, it is not behaving well. I think the test is not enough to validate the correctness of Jacobian.

eastskykang, 2021-03-22:
As we mentioned, you can consider the test as a baseline. Passing all test doesn't mean you get a full point from the exercise neither your implementation is correct.

If you want to extend your test to more general cases. Copy paste the provided test code and please write more test cases in your own test code file.

Please consider that the debugging process is also part of this assignment. We hope you learn not only theories behind but also how to implement your logic in C++ code.

Catially, 2021-03-22:
Sure! Thanks, Dongho!

closed 2021-03-22 by Catially

eastskykang, 2021-03-22:
@Catially nonetheless, I appreciate for your report. I think this could be helpful to your colleague students.

eastskykang, 2021-03-29:
@Catially can you tell me if you fixed your problem? Was it somehow related to #33?

Catially, 2021-03-29:
> @Catially can you tell me if you fixed your problem? Was it somehow related to #33?

Yes I have fixed it. The reason of this issue is my wrong implementation of analytical Jacobian. 😂