cmm-21/a2 #15: Downloadable tutorial recordings?

AntaresZhang, 2021-03-23:
Dear TA: 

Will it be possible to release the tutorial videos in the same (downloadable) way as the lecture videos?

it will be easier when we have network speed issues especially trying to access from outside Switzerland.

Many Thanks!

Best,

Antares

eastskykang, 2021-03-23:
We uploaded tutorial video to our course website. Please check it out. 

We are not sure if we will upload lecture recordings though. I will discuss with TAs and professors.

AntaresZhang, 2021-03-23:
Dear TA:
Great thanks for your help! 
I checked again and was able to download the Tutorial 2 video through the direct link.

For lecture videos, I think the current links are already downloadable.

Thanks again! I will close this issue now

Antares

closed 2021-03-23 by AntaresZhang