cmm-21/a2 #16: Calculation of Analytic Jacobian

PanagiotisAnagnostaras, 2021-03-23:
Hey everyone,
 I try to do the analytic jacobian but I have some questions:
[image]

Don't we need to know the lengths of the links and which q corresponds to each joint, or obtaining this info from the provided code is part of the assignment with reverse engineering?

eastskykang, 2021-03-23:
Please do not hard code length of the link. Your implementation should be general to use "any" kind of kinematic tree.

You can access position of joint w.r.t. parent rigid body's coordinate frame by ```joint->pJPos```and position of joint w.r.t. child rigid body's coordinate frame by ```joint->cJPos```

eastskykang, 2021-03-25:
More detail instruction was given in the last tutorial session. It will be uploaded to the course website shortly. 

Hope this helps!

closed 2021-03-25 by eastskykang