cmm-21/a2 #18: Ex3 video recording

iqiac, 2021-03-24:
I managed to solve Exercise 3, but when I play it, the FPS drops from 30 to around 5 and the robot moves very slow. To show everything, it takes me a bit less than a minute. Is this related to something in the code or is it fine if I just edit the video s.t. it plays faster to get about 10sec?

eastskykang, 2021-03-24:
@iqiac please make sure if you are building your code in cmake Release mode. (give the following cmake option flag when you run cmake)

```sh
$ cmake -DCMAKE_BUILD_TYPE=Release ../
```

eastskykang, 2021-03-24:
IK is not a computationally heavy operation. It's not necessarily to run in 30 fps (real time) but if it's too slow, something could be wrong.

iqiac, 2021-03-24:
I'm using vs code on windows. The only file with cmake -DCMAKE_BUILD_TYPE=Release is in "./build/_deps/eigen-src/scripts/release.in" where it uses "." and not "../". Changing to "../" didn't do anything, it still drops to 5-6 FPS (link: https://www.youtube.com/watch?v=Kxig2WnUob8). So you think this is a mistake in the IK implementation?

relazzouzi, 2021-03-24:
I have the same issue as well, as soon as I press play it drops to 3-4 FPS, making it impossible to show what the robot can do within just 10 seconds. Also using VS code on windows.

eastskykang, 2021-03-24:
[image]

Please tell me what is your cmake build mode from the status bar here. It should be "Release", not "Debug" if you want to build with optimization in execution speed.

iqiac, 2021-03-24:
Ah thank you, it was on debug, I changed to release and it doesn't drop the framerate now. Thanks very much.

closed 2021-03-24 by iqiac