cmm-21/a2 #26: Robot Model start in ground

Maenzgi, 2021-03-25:
Dear TAs and fellow students

I am having a weird issue, and I am not sure if it is my fault or if it is a bug in the program structure: I think I've implemented Task 1 correctly, and the green spheres appear where they should (if I compare them to the images in the readme file). However, for some reason the robot is in the ground. I haven't changed anything in the code except in the two functions that we have to work on in task 1.
[image]

iqiac, 2021-03-25:
see #6

Maenzgi, 2021-03-25:
> 
> 
> see #6

I think it is not the same issue, because my spheres are at the correct position, and only the robot is in the ground. In #6, everything is shifted

iqiac, 2021-03-25:
Ah sorry. In that case, it looks like something went wrong with maybe the last translation step where you add the x_root (kinematics slide 39). I don't have further ideas at the moment.

eastskykang, 2021-03-25:
As @iqiac suggested, please see #6 and set starting height to be 0.5.

eastskykang, 2021-03-25:
BTW somehow I see the green spheres are not located in correct position. Please do a double check your forward kinematics implementation.

closed 2021-03-26 by Maenzgi