cmm-21/a2 #28: Base Trajectory discontinous on the boundry of uneven terrain

Hustwireless, 2021-03-25:
Hi,

I've set the boundry of the sphere shape bump based on calculation, however, the base trajectory appears to be discontinous on the boundry of the uneven terrain. Has anyone encountered this problem?

I can certainly make the boundry larger to eliminate this discontinouity, however it will not be consistent with my calculation... Is this also due to the inaccuracy of the meshes?

Thanks! 

[image]

eastskykang, 2021-03-25:
This shouldn't be the case in my opinion. Please do a double check your implementation. 

BTW, as I mentioned in the tutorial session, you don't need to generate trajectory over a terrain. Giving offset to IK targets are enough.

Hustwireless, 2021-03-25:
Thanks! I'll check that and see if I can implement by giving offset to IK targets. BTW, my sphere looks more like a polygon here... I don't know how to make it more round...

eastskykang, 2021-03-25:
@Hustwireless this is very normal. You don't need to worry about that. (one trick is using sphere mesh with high resolution. but this is not required.)

lthiet, 2021-03-27:
Hey, I had the same problem and found a solution, but I think I cannot outright give you the answer, but I'm assuming that you are taking the norm of the target position in the trajectory planer to see when you should add the offset. Check again whose norm you should use to trigger the sphere calculation. That solved it for me.

Hustwireless, 2021-03-28:
> Hey, I had the same problem and found a solution, but I think I cannot outright give you the answer, but I'm assuming that you are taking the norm of the target position in the trajectory planer to [...]

Hi! Thanks for the information! However, I took the norm of the distance from the base pos to the xz coordinate of the center of the sphere, is it already the norm that you suggested, or something else?

lthiet, 2021-03-28:
Did you also only consider the xz coordinate in the base pos? For me I considered all xyz coordinates and removing the y coordinates solved the trajectory not being continuous around the boundary of the sphere

Hustwireless, 2021-03-29:
Thanks a lot! @lthiet  It works!

eastskykang, 2021-03-29:
@Hustwireless we don't consider joint angle limit in our IK formulation. So it's very natural that the leg snaps into other direction (in near-singular configuration). Of course this doesn't really look good, so when you record a demo video, try not to let such configuration happens.