cmm-21/a2 #32: fps too low

boduan1, 2021-03-28:
I use linux, and I am already in released mode, but it fps is 7.
[image]
[image]

eastskykang, 2021-03-28:
I need more information. Does it drop as you play the app? Or is it already slow when you launch the app? 

(Please be more specific. We cannot reproduce or figure out your problem only with the piece of information you provided here)

boduan1, 2021-03-28:
It doesn't drop, and I tried both command line (adding -DCMAKE_BUILD_TYPE=Release in cmake)
and directly use VScode release mode, both gives 7 fps...
Maybe it is just because my computer is slow?

eastskykang, 2021-03-29:
I quickly tested your code, and could run your code in real-time. Could you post the system info of your PC? You can go to settings > details.

e.g. 
[image]

boduan1, 2021-03-29:
Thanks! I tried to move to another windows computer, and it works on windows, but the linux system is still slow...
[image]

eastskykang, 2021-03-29:
strange. I think your CPU is powerful enough to run the IK solver in real time. 

Can you delete your build directory completely and configure, build your project from the scratch? If it's still slow, might be related to graphics.

Hustwireless, 2021-03-29:
Could it be that you are using fractional scale on a high resolution screen?

boduan1, 2021-03-29:
Oh! that might be the problem! the screen resolution is high...

closed 2021-03-29 by boduan1

yuliangzhong, 2021-03-31:
I have the same problem... even worse! My FPS is only 2-3. I want to know if my code works when you testing... 

> I quickly tested your code, and could run your code in real-time on my laptop (macOS 2.4 GHz 8-Core Intel Core i9.) Could you post the system info of your PC? You can go to settings > details.
> 
> e.g.
> [image]

eastskykang, 2021-04-02:
If your FPS is too low as soon as you launch the app:

1. double check if your CMake mode is Release 
2. try to lower your screen resolution