cmm-21/a2 #35: [Announcement] Complex terrain for extra 5% point.

eastskykang, 2021-03-29:
If you are already done with all exercises (and also bored), please try to implement exercise 5 for a more complex terrain loaded from a mesh file. If you manage to successfully implement this, you will get extra 5% point for this assignment (but your final point cannot exceed 100%)

This could be pretty complicated and may require a lot of code modifications. **So please create a new branch called ```meshterrain``` and work there while keep your successful implementation of Ex.1~Ex.5 on the master (or main) branch.** Again, please make sure your implementation of Ex.1 ~ Ex.5 stays safe in master (or main) branch. 

Successful implementation will looks like this:

[image]

[video](https://www.youtube.com/watch?v=EYh9Po8piIM)

You can download a terrain file you want to use from any internet source, but if it's hard to find one, you can use one in data directory ```data/terrain/terrain.obj``` 

Note. this will require you to read our code base thouroughly. 

Let me give some hint.  

1. You can create a mesh file by 

```cpp
// for instance
Model terrain = Model(CRL_DATA_FOLDER "/terrain/terrain.obj");
```

2. You can draw a mesh by 

```cpp
// in app.h
virtual void drawObjectsWithShadows() override {
    // TODO: comment this line
    // ground.draw(shadowShader, V3D(0.6, 0.6, 0.8));

    // TODO: and add this
    terrain.draw(shadowShader);
    robot->draw(shadowShader);
}
```

3. You can use the following function for getting a ground height 

```cpp
// member function of Model class
bool Model::hitByRay(const P3D &r_o, const V3D &r_v, P3D &hitPoint) const;
```

the strategy is as follows. From underground e.g. (x, -1.0, z), we will shoot a ray toward the sky (straight up) and then will find a ```hitPoint```. The ```hitPoint.y```  is the ground height at (x, z) coordinate.

Hustwireless, 2021-03-29:
Thanks for the instructions!

iqiac, 2021-03-30:
[image]
I managed to make the robot walk on uneven terrain. However thes "lines" have still the old height. Does this influence the points negatively?

eastskykang, 2021-03-30:
@iqiac this is fine. Looks great. 

Please just try to make the demo video visible enough so that we can clearly see the robot walks on the uneven terrain.

kevinhangoat, 2021-04-02:
When I tried to call `SimpleGroundModel terrain` in files in simAndControl, I go this error:

/home/kevin/ETHz/CMM/a2-kevinhangoat/src/libs/simAndControl/kinematics/IK_Solver.h:31:9: error: ‘SimpleGroundModel’ was not declared in this scope
         SimpleGroundModel terrain;

Does this mean I need to change cmake file? Or I am supposed to do something different?

eastskykang, 2021-04-02:
@kevinhangoat You don't need to change anything from cmake side for this exercise. 

FYI. You should not use SimpleGroundModel for a mesh terrain. SimpleGroundModel is a flat ground. Please use hints I wrote above.

kevinhangoat, 2021-04-02:
> @kevinhangoat You don't need to change anything from cmake side for this exercise.
> 
> FYI. You should not use SimpleGroundModel for a mesh terrain. SimpleGroundModel is a flat ground. Please use hints I wrote above.

Thanks! But still I do not understand why I cannot call class Model in files like IK_Solver. So I am not supposed to go down that track?

eastskykang, 2021-04-02:
@kevinhangoat  Did you add the header file to IK_Sover.h?

```cpp
#include <gui/model.h>
```

kevinhangoat, 2021-04-02:
@eastskykang Yes, I did. The full error is:
```
/home/kevin/ETHz/CMM/a2-kevinhangoat/src/libs/simAndControl/kinematics/IK_Solver.h: In member function ‘void crl::IK_Solver::addEndEffectorTarget(crl::RB*, crl::P3D, crl::P3D)’:
/home/kevin/ETHz/CMM/a2-kevinhangoat/src/libs/simAndControl/kinematics/IK_Solver.h:31:9: error: ‘Model’ was not declared in this scope
         Model terrain = Model(CRL_DATA_FOLDER "/terrain/terrain.obj");
         ^~~~~
/home/kevin/ETHz/CMM/a2-kevinhangoat/src/libs/simAndControl/kinematics/IK_Solver.h:31:9: note: suggested alternative:
In file included from /home/kevin/ETHz/CMM/a2-kevinhangoat/src/libs/gui/include/gui/renderer.h:6:0,
                 from /home/kevin/ETHz/CMM/a2-kevinhangoat/src/app/app.h:6,
                 from /home/kevin/ETHz/CMM/a2-kevinhangoat/src/app/main.cpp:3:
/home/kevin/ETHz/CMM/a2-kevinhangoat/src/libs/gui/include/gui/model.h:10:7: note:   ‘crl::gui::Model’
 class Model {
       ^~~~~
src/app/CMakeFiles/locomotion.dir/build.make:81: recipe for target 'src/app/CMakeFiles/locomotion.dir/main.cpp.o' failed
make[2]: *** [src/app/CMakeFiles/locomotion.dir/main.cpp.o] Error 1
CMakeFiles/Makefile2:687: recipe for target 'src/app/CMakeFiles/locomotion.dir/all' failed
make[1]: *** [src/app/CMakeFiles/locomotion.dir/all] Error 2
Makefile:159: recipe for target 'all' failed
make: *** [all] Error 2

```

eastskykang, 2021-04-02:
please try ```gui::Model``` instead of ```Model```

kevinhangoat, 2021-04-02:
@eastskykang Thank you so much, it works! I still have one problem that FPS is quite low, although I am already in release mode. What could be the problem?

eastskykang, 2021-04-02:
@kevinhangoat does FPS drops when you play the app (by pressing space bar)? or is it already low as soon as you launch the app?

kevinhangoat, 2021-04-02:
@eastskykang It drops when it starts playing. It happens if I load the mesh terrain and start playing (without any other change). Maybe my linux system is just slow?

eastskykang, 2021-04-02:
Oh... it's possible. Mesh rendering is a bit heavy computation.

Rasilu, 2021-04-02:
Maybe you are loading the terrain inside a loop? That was what caused most FPS drops for me.