cmm-21/a2 #37: Ex. 3 Trajectory Planning

nwicki, 2021-04-01:
Hi

I am currently doing task 3 and it looks pretty good. My robot is able to walk back and forth, left and right and in different speeds with no problem (I made the direction dependent on the heading, the time step, and the inputs given by the user. If this is not planned to be like this, please tell me.)

I am having troubles though adjusting the heading angle. As before I made the updated heading angle dependent on the turning speed and the time step and the robot is able to walk in that direction, but it flaps its legs around wildly and I am not sure if it is caused by a wrong update of the heading angle or if this is caused by a previous implementation error (the IK and FK solver seem pretty straight forward and since I get reasonable results, I am doubting to have made any mistakes there).

The only thing I haven't yet added is regularisation and line search, but I cannot see how this might affect the movement of the legs.

Any insight is appreciated :)

eastskykang, 2021-04-01:
Could you post a short demo video here so that we can understand better what happens?

nwicki, 2021-04-01:
Yes, you can watch it here: https://youtu.be/b000ewjF3lU

eastskykang, 2021-04-01:
Oops.... this is not good. I think something wrong in your implementation 

Please quickly go through the following issues:
- https://github.com/cmm-21/a2/issues/12

and do the double check if your implementation is correct.

eastskykang, 2021-04-01:
[image]

This stillshot indicate your forward kinematics implementation is not correct, (green spheres are not around the feet anymore)

I suspect it does not properly handle the base orientation.

nwicki, 2021-04-01:
Indeed, issue #8 was very helpful. Thanks for pointing me to it.

I forgot to handle the outermost case of the forward kinematics recursion for the root where I did add the base, but forgot to handle the orientation. Since it looked alright with no orientation change I assumed it to be correct.

Thanks again.

closed 2021-04-01 by nwicki