cmm-21/a2 #38: Ex 2-2 not converging

swellsc, 2021-04-01:
Hi, I am having issues with ex. 2-2 and I cannot manage to pinpoint where the issue is arising from. The robot trajectory is diverging as is shown in the following image. I am using Gauss-Newton method and inverting using ldlt().solve() and have followed all the advice found here on github issues.
[image]
I have tried adding a regularizer to the method and the trajectory first diverges as above and after some steps converges to the desired end-effector trajectory, but with a different joint configuration than desired, with the knees backwards:
[image]
I am wondering if anyone has had any similar issue and, above everything, where the issue might be coming from, as I have checked everything again and again and cannot find what is wrong in my code. Any help is appreciated!
Thanks a lot!

eastskykang, 2021-04-01:
If your Gauss-Newton as well as Jacobian implementation is correct, it should be stable enough. It's a bit suspicious that it becomes unstable as soon as you play the app. 

Again, trajectory is not an issue here. The problem arises from either IK or Jacobian.

swellsc, 2021-04-01:
Indeed there was a small issue in my IK implementation that I just solved. Thank you!

Now it is stable and does not need a regularizer. However, the frame rate I get is about 10 fps. Is this expected? Or is there still some efficiency or other issue I should solve in my code?

eastskykang, 2021-04-01:
@swellsc please see the issue #4 and try to change your Cmake mode into Release mode. If it's already Release mode, it might be related to resolution of your display. (if it's too high res, frame rate could be low) If that's the case, please lower the resolution settings of your display and try it again.

swellsc, 2021-04-01:
Changing cmake mode to Release solved it. All works smoothly now, thanks for all your help!

closed 2021-04-01 by swellsc