cmm-21/a2 #39: github check fails but no errors were seen

kevinhangoat, 2021-04-02:
Hello, I noticed that when I pushed, github checks fail, showing:
```
📝 Test-a2

  -- The C compiler identification is GNU 9.3.0
  -- The CXX compiler identification is GNU 9.3.0
  -- Detecting C compiler ABI info
  -- Detecting C compiler ABI info - done
  -- Check for working C compiler: /usr/bin/cc - skipped
  -- Detecting C compile features
  -- Detecting C compile features - done
  -- Detecting CXX compiler ABI info
  -- Detecting CXX compiler ABI info - done
  -- Check for working CXX compiler: /usr/bin/c++ - skipped
  -- Detecting CXX compile features
  -- Detecting CXX compile features - done
  -- Configuring done
CMake Error at CMakeLists.txt:34 (add_library):
    Target "simAndControl" links to target "crl::gui" but the target was not
    found.  Perhaps a find_package() call is missing for an IMPORTED target, or
    an ALIAS target is missing?
  Call Stack (most recent call first):
    src/libs/simAndControl/CMakeLists.txt:54 (create_crl_library)
  
  
  CMake Error at src/test-a2/CMakeLists.txt:5 (add_executable):
    Target "test-a2" links to target "crl::gui" but the target was not found.
    Perhaps a find_package() call is missing for an IMPORTED target, or an
    ALIAS target is missing?
  
  
  CMake Error at CMakeLists.txt:36 (add_dependencies):
    The dependency target "crl::gui" of target "simAndControl" does not exist.
  Call Stack (most recent call first):
    src/libs/simAndControl/CMakeLists.txt:54 (create_crl_library)
  
  
  -- Generating done
  CMake Generate step failed.  Build files cannot be regenerated correctly.
  
❌ Test-a2
::error::Error: Exit with code: 1 and signal: null

::7b62a92b-d290-40ce-85ed-e58bf2810f99::
```

I do not understand why the cmake files fail. I did not see any bugs when running locally and I did pass test-a2. Thanks

deltahue, 2021-04-02:
I am not a TA so no guarantees, but I think this does not matter (see https://github.com/cmm-21/a2/issues/3)

kevinhangoat, 2021-04-02:
I see, Thank you!

closed 2021-04-02 by kevinhangoat